sparkfun/Arduino_Apollo3 #443: unique chip ID accessible from the sketch?

jerabaul29, 2022-01-14:
I was wondering about the following: is there some form of "unique" chip ID that could be use as an ID tag? For example, a serial number "burnt" somehow on the Ambiq chip that can be read by a program and used further? Or maybe (even if it is not stricly "unique" as the bootloader can be erased and re-written), a "unique" bootloader ID number or similar?

Would be very helpful to help track actual boards when these should for example receive some config that may need to be re-uploaded if the board gets a full reset or so (thinking about having a small fleet of boards + 9dof, and the question of the 9dof soft and hard iron compensation; I would like to calibrate each board only once, and be sure in the future that I can easily identify which board corresponds to which calibration; the best way to do so would be that each board has an ID that is constant across reboots / uploading of new firmware / etc).

paulvha, 2022-01-15:
you can read the unique CHIPID. (it is actually used to set a unique address for BLE)

`void setup() {
  
  Serial.begin(115200);
  Serial.println("\nGet ID number ");

  // for more details look at am_hal_mcuctrl.h
  am_hal_mcuctrl_device_t sDevice;
  am_hal_mcuctrl_info_get(AM_HAL_MCUCTRL_INFO_DEVICEID, &sDevice);

  Serial.print("ChipID0 : 0x");
  Serial.println(sDevice.ui32ChipID0, HEX);

  Serial.print("ChipID1 : 0x");
  Serial.println(sDevice.ui32ChipID1, HEX);
}

void loop() {
  // put your main code here, to run repeatedly:

}`

jerabaul29, 2022-02-04:
That works very well, many thanks :) . Wonder if this could be added somewhere in some documentation / readme / example sketch :) . Feel free to close.

closed 2022-03-17 by Wenn0101

Wenn0101, 2022-03-17:
Updated example 10 to include this in the example