sparkfun/Arduino_Apollo3 #444: About MBedOS: are you using v6? possible to use with both RTOS and BareMetal

jerabaul29, 2022-01-14:
A disclaimer first: I am myself still only using the core v1 (ie arduino "bare metal" core). I had tested the MBedOS core back in January 2021, and too much stuff did not work, there was too much overhead, things were too slow etc. I am happy with the bare metal core v1, and I do not plan to move old projects to the MBedOS version, but I am still open to thinking about all alternatives when it regards new projects.

My (renewed?) interest in the MBedOS core is that if I understand well, following the release of MBedOS v6, MBedOS now comes in two flavors with inter-operable APIs: v6 RTOS and v6 BareMetal (see https://os.mbed.com/docs/mbed-os/v6.15/introduction/index.html ).

My question is therefore:

- which MBedOS version is used in the MBedOS core? Is it up to v6?
- is the MBedOS core, in case it is in v6, able to use both the RTOS and the BareMetal flavors?

I spent / "lost" a lot of time looking at MBedOS a year ago, so I want to understand things better before I consider sinking time into it again, especially when I am mostly very happy with the core v1 bare metal...

Wenn0101, 2022-01-18:
The arduino core is using v5 of MBed still. The boards are all valid MBed6 targets, but we havent had significant reason to upgrade yet (and risk breaking peoples projects).

The v2 core is only a RTOS implementation.

jerabaul29, 2022-01-18:
Ok, thanks. Would be great to have it explained in the Readme, as these MBedOS versions get quite confusing, and it has quite a bit to say for which APIs are available or not, etc.

If you ever move to MBedOS6 that would be amazing, as this may be a path towards having a unified API for both RTOS and bare metal cores :) .

Wenn0101, 2022-03-17:
Mbed6 has been on the list for a little while. There are a couple things it would bring, but so far the benefits don't seem to outweigh the risks of potentially breaking people projects. I will let you know if/when we update the core. Mbed5 also has some "Baremetal" options, but they require a recompile of the mbed core underneath, and leave out many of the features that people want.

Mbed versions are confusing, but if you curious what version of mbed we are using, it is our fork of the [repo](https://github.com/sparkfun/mbed-os-ambiq-apollo3).

closed 2022-03-17 by Wenn0101