sparkfun/Arduino_Apollo3 #451: Deep sleep not working with BLE

Brezensalzer, 2022-03-05:
Hi,
when BLE is enabled ( `BLE.begin()` ) the sleep modes won't work anymore.
`am_hal_sysctrl_sleep(AM_HAL_SYSCTRL_SLEEP_DEEP)`
or
`am_hal_sysctrl_sleep(AM_HAL_SYSCTRL_SLEEP_NORMAL)`
have no effect.

Arduino IDE: 1.8.19
Apollo 3 core: 2.2.0
Board: RedBoard Artemis Nano

paulvha, 2022-03-06:
The library is NOT Sparkfun, nor maintained by Sparkfun. Issues should be posted here https://github.com/arduino-libraries/ArduinoBLE/issues. There are a number of serious concerns mentioned there that they do not solve.  Some time ago I did a longer deep dive on ArduinoBLE. I have fixed a large number of issues for myself, including the one you encounter.  I have posted the issue, the root cause and solution over a year ago (https://github.com/arduino-libraries/ArduinoBLE/issues/158) with the changes to be made. Not sure why they do not apply the fix but you can apply it yourself.

regards,
Paul

Brezensalzer, 2022-03-06:
Thanks for the link, I will have a look there!

Wenn0101, 2022-03-17:
Closing. Re-open this issue if you don't get the answers you are looking for from the team maintaining ArduinoBLE.

closed 2022-03-17 by Wenn0101