sparkfun/Arduino_Apollo3 #453: Add support for Arduino CLI/IDE 2.0 Debugging

marsfan, 2022-03-19:
The Arduino CLI and Upcoming Arduino IDE 2.0 support actual hardware debugging. They have support for a number of different JTAG and SWD Debuggers. It would be nice to add support for using this new debugging capability with the Artemis boards.

The documentation on supporting this is fairly minimal at the moment, but I found some info here; https://arduino.github.io/arduino-cli/0.21/platform-specification/#sketch-debugging-configuration
As the **Nano 33 IoT** currently supports debugging, it might be possible to look at its board definition files to figure out how to add support.

Wenn0101, 2022-03-23:
I will look into this. If I recall correctly, in order to cut down the package size, we removed the debugging symbols from the precompiled mbed core. Perhaps I will have to create a second build with these added back in.

Either way this is definitely a feature worth looking into.

paulvha, 2022-03-23:
Actually, this is different. The debug tool / option in Arduino IDE 2.0 is only available for boards that have both a separate interface 'chip' and a target 'chip'. A board with a SAMD (as within the Nano) has both in a single chip. It can be used with Daplink or UF2 and a virtual filesystem as such the only Artemis board is the DK. Good articles about this are https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-debugger and  https://makecode.com/blog/one-chip-to-flash-them-all

regards,
Paul