sparkfun/Arduino_Apollo3 #461: Failing to upload to Sparkfun Edge board using Arduino IDE 1.8.19

asanjib28, 2022-04-23:
I struggled to upload a simple blink code into a Sparkfun Edge board, for the past month, but did not succeed yet. In fact, I purchased a 2nd board as well thinking that I might have damaged the first board. But the second also behaved the same. I am using a Sparkfun recommended USB-C to UART (3.3V) breakout, which uses CH340C. My computer has Windows 10 and I have tried both Arduino IDE 1.8.19 version and Arduino IDE 2.0.0-rc6. But no success.

When I use Arduino IDE 1.8.19, my settings are:
[image]

Then when I press the upload button, it tries three times and then fails. I tried reducing the baud rate. I tried pressing buttons as mentioned in the Sparkfun tutorial. Again I have also tried not touching any buttons. I could see that the Tx LED on the CH340C breakout board blinks momentarily but Rx does not blink. The edge board gets reset when the Tx LED blinks (if I do not keep the button 14 pressed on), and it does three times before declaring fail to upload: 
================================================================
Header Size =  0x80
original app_size  0x2224 ( 8740 )
load_address  0xc000 ( 49152 )
app_size  0x2224 ( 8740 )
w0 = 0xcb0022a4
Security Value  0x10
w2 =  0x10008080
addrWord =  0xc000
versionKeyWord =  0x0
child0/feature =  0xffffffff
child1 =  0xffffffff
crc =   0xf331a6c3
Writing to file  C:\Users\Sanjib\AppData\Local\Temp\arduino_build_222327/Blink_SparKFunEdge.ino_OTA_blob.bin
testing: C:\Users\Sanjib\AppData\Local\Temp\arduino_build_222327/Blink_SparKFunEdge.ino_OTA_blob.bin
Header Size =  0x60
app_size  0x22a4 ( 8868 )
Writing to file  C:\Users\Sanjib\AppData\Local\Temp\arduino_build_222327/Blink_SparKFunEdge.ino_Wired_OTA_blob.bin
Image from  0x0  to  0x22a4  will be loaded at 0x20000
Connecting over serial port COM16...
Sending Hello.
No response for command 0x00000000
Failed to respond
Fail
Sending Hello.
No response for command 0x00000000
Failed to respond
Fail
Sending Hello.
No response for command 0x00000000
Failed to respond
Fail
Tries = 3
Upload failed
=======================================================================

I am exhausted trying. Any lights, please....

Regards,
Sanjib

paulvha, 2022-04-23:
The button order is very important. The hookup guide looks different from what I do :

    Press and hold the Reset button
    While still holding down the Reset button, press and hold the 14 button
    Release the Reset Button
    **KEEP THE 14 BUTTON PRESSED**  DO NOT  : Release the 14 button !!
    Hit Upload in the Arduino IDE (or use your keyboard shortcut)

GIve it a try.
regards,
Paul

asanjib28, 2022-04-23:
@paulvha, it works!! Great! I followed the sequence earlier, but as it is very hard to keep the button pressed for a longer duration, I might have released button 14 some point of time after releasing RESET button. This time I ensured I keep the button pressed till the upload is completed and it works on both the boards I have. Thanks a lot!!! I owe you one!!!

Regards,
Sanjib

paulvha, 2022-04-23:
Good to hear it worked.

You have to keep it pressed until it starts to upload. Then you can release it.
@Wenn0101,  can you pass this on to someone who can upload the hookup page for the edge ?

regards,
Paul

hhuhuhnjzy, 2025-03-22:
this worked very well.thanks thanks thanks