sparkfun/Arduino_Apollo3 #470: Missing core v1 support for Sparkfun LoRa Thing Plus expLoRaBLE

larswd, 2022-08-30:
@jerabaul29 and I are working on coding some sensors using the "Sparkfun LoRa Thing Plus expLoRaBLE" as a basis. 
Unfortunately, these boards are seemingly only supported on the v2 core, and does not show in the board manager for the Arduino IDE when using the core v1.  
For our purposes, we need the v1 core. Could it be possible to add support for this board in core v1 as well?

paulvha, 2022-08-31:
The big(gest) issue of course is that BLE is NOT supported on V1 by Sparkfun. Looking at the different source codes, there is not much that would stop it on V1. Maybe some defines...

But (just out of interest and you can also drop me a personal mail on paulvha@hotmail.com) what is blocking you from using V2?

regards
Paul

jerabaul29, 2022-08-31:
- I was "traumatized" by working with the core v2 in late 2020, it was a big mess, but I agree the core v2 may be much better now...
- when trying again the core v2 more recently, I bumped on a number of issues, for example something similar to what is described at: https://github.com/paulvha/apollo3/blob/master/OneWire_on_uart/1-wire_uart.odt
- larger overhead; a problem when I need to use ~60% of the RAM for storing data before taking a series of FFT

But if it is unrealistic to use core v1 with LoRa, guess @larswd could give a try at the core v2.

paulvha, 2022-09-01:
Agree with your points. Indeed in V2 the overhead (mainly caused by the inclusion of Mbed) brings extra challenges. Not much can be done about that. Maybe in this project from V2 is a good fit.

regards,
Paul