sparkfun/Arduino_Apollo3 #472: interrupts() and noInterrupts() do not have any definition.

bimalpaneru, 2022-11-14:
Without interrupts() and noInterrupts()  there is compilation error Arduino LMIC library, Hard to have the RF95 module get working

Leenix, 2022-11-30:
The definitions for `interrupts` and `noInterrupts` are as below (from [arduino-mbed-bridge/core-extend/Common.h](https://github.com/sparkfun/arduino-mbed-bridge/blob/25fe1981fe509217059f093395372e1ffcbd6335/core-extend/Common.h)):

```
#define interrupts() (void)am_hal_interrupt_master_disable()
#define noInterrupts() (void) am_hal_interrupt_master_enable()
```

Does that seem backwards to you?

`interrupts` should globally enable interrupts and `noInterrupts` globally disable them.

I couldn't find the specific code to see what's going on with the registers, but the words and the actions don't seem to match.

paulvha, 2022-11-30:
That is what they do. 

They are defined in SparkFun/hardware/apollo3/2.2.1/cores/arduino/mbed-bridge/core-extend/common.h (line 54)

Leenix, 2022-12-01:
I am observing the opposite effects.

`interrupts()` should be defined with `am_hal_interrupt_master_enable()` to enable interrupts.
It is currently defined with `am_hal_interrupt_master_disable()`.

The definitions are incorrect and should be switched.

Leenix, 2022-12-01:
@paulvha 
Also, I couldn't follow your pathing in this repo.

I'm assuming it's the same file that I linked above, which contains the definitions, but line 54 refers to `indexTone`. 
Are we looking at the same file?

paulvha, 2022-12-01:
you are correct that it works the opposite : 

#define interrupts() (void)am_hal_interrupt_master_disable()
#define noInterrupts() (void) am_hal_interrupt_master_enable()

As it related to the line 54.. I have made changes to that file. The original version 2.2.1, same file, but lines 40 and 41

nigelb, 2024-03-26:
Hi @paulvha, if I checkout the most recent release `v2.2.2`:

```bash
$ git clone --recurse-submodules --branch v2.2.2 --depth 1 https://github.com/sparkfun/Arduino_Apollo3.git
Cloning into 'Arduino_Apollo3'...
remote: Enumerating objects: 280, done.
remote: Counting objects: 100% (280/280), done.
remote: Compressing objects: 100% (232/232), done.
remote: Total 280 (delta 83), reused 139 (delta 34), pack-reused 0
Receiving objects: 100% (280/280), 2.63 MiB | 2.03 MiB/s, done.
Resolving deltas: 100% (83/83), done.
Note: switching to '9f3dc4d88e9a42f262c15b756c82e6beb4897b6f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Submodule 'cores/arduino/mbed-bridge' (https://github.com/sparkfun/arduino-mbed-bridge) registered for path 'cores/arduino/mbed-bridge'
Submodule 'cores/mbed-os' (https://github.com/sparkfun/mbed-os-ambiq-apollo3) registered for path 'cores/mbed-os'
Submodule 'tools/keywords_gen' (https://github.com/sparkfun/Keyword_File_Generator) registered for path 'tools/keywords_gen'
Submodule 'tools/uploaders/asb' (https://github.com/sparkfun/Apollo3_Uploader_ASB) registered for path 'tools/uploaders/asb'
Submodule 'tools/uploaders/svl' (https://github.com/sparkfun/Apollo3_Uploader_SVL) registered for path 'tools/uploaders/svl'
Cloning into '/home/user/Documents/Arduino_Apollo3/cores/arduino/mbed-bridge'...
remote: Enumerating objects: 199, done.        
remote: Counting objects: 100% (199/199), done.        
remote: Compressing objects: 100% (116/116), done.        
remote: Total 199 (delta 118), reused 149 (delta 70), pack-reused 0        
Receiving objects: 100% (199/199), 34.99 KiB | 1.09 MiB/s, done.
Resolving deltas: 100% (118/118), done.
Cloning into '/home/user/Documents/Arduino_Apollo3/cores/mbed-os'...
remote: Enumerating objects: 321427, done.        
remote: Counting objects: 100% (36/36), done.        
remote: Compressing objects: 100% (36/36), done.        
remote: Total 321427 (delta 0), reused 29 (delta 0), pack-reused 321391        
Receiving objects: 100% (321427/321427), 386.86 MiB | 12.27 MiB/s, done.
Resolving deltas: 100% (223002/223002), done.
Cloning into '/home/user/Documents/Arduino_Apollo3/tools/keywords_gen'...
remote: Enumerating objects: 62, done.        
remote: Counting objects: 100% (4/4), done.        
remote: Compressing objects: 100% (4/4), done.        
remote: Total 62 (delta 0), reused 1 (delta 0), pack-reused 58        
Receiving objects: 100% (62/62), 742.58 KiB | 4.15 MiB/s, done.
Resolving deltas: 100% (26/26), done.
Cloning into '/home/user/Documents/Arduino_Apollo3/tools/uploaders/asb'...
remote: Enumerating objects: 574, done.        
remote: Counting objects: 100% (16/16), done.        
remote: Compressing objects: 100% (12/12), done.        
remote: Total 574 (delta 2), reused 12 (delta 1), pack-reused 558        
Receiving objects: 100% (574/574), 149.29 MiB | 11.91 MiB/s, done.
Resolving deltas: 100% (235/235), done.
Cloning into '/home/user/Documents/Arduino_Apollo3/tools/uploaders/svl'...
remote: Enumerating objects: 676, done.        
remote: Counting objects: 100% (201/201), done.        
remote: Compressing objects: 100% (82/82), done.        
remote: Total 676 (delta 94), reused 184 (delta 83), pack-reused 475        
Receiving objects: 100% (676/676), 204.96 MiB | 12.39 MiB/s, done.
Resolving deltas: 100% (285/285), done.
remote: Enumerating objects: 64, done.
remote: Counting objects: 100% (64/64), done.
[... 18 lines cut ...]
Unpacking objects: 100% (30/30), 9.31 KiB | 3.10 MiB/s, done.
From https://github.com/sparkfun/Keyword_File_Generator
 * branch            483ce55047a2adb3210eddd3bff2a05f658a4373 -> FETCH_HEAD
Submodule path 'tools/keywords_gen': checked out '483ce55047a2adb3210eddd3bff2a05f658a4373'
Submodule path 'tools/uploaders/asb': checked out '454fc619ce9371016f7bbdbb875aed2e197ea1ce'
Submodule path 'tools/uploaders/svl': checked out '54a37d5009fd8bb4e5c9770cabd4bed984ac7c98'
```
Then have a look at the submodules:
```git
$ git submodule
 25fe1981fe509217059f093395372e1ffcbd6335 cores/arduino/mbed-bridge (heads/master-5-g25fe198)
 50912b669e4b9ba484d534484190a4a362926c2f cores/mbed-os (mbed-os-5.4.0-rc1-20343-g50912b669e)
 483ce55047a2adb3210eddd3bff2a05f658a4373 tools/keywords_gen (v1.0.0-8-g483ce55)
 454fc619ce9371016f7bbdbb875aed2e197ea1ce tools/uploaders/asb (heads/master)
 54a37d5009fd8bb4e5c9770cabd4bed984ac7c98 tools/uploaders/svl (54a37d5)
```
We find that the `cores/arduino/mbed-bridge` submodule is at hash `25fe1981fe509217059f093395372e1ffcbd6335`.
If we look at the network graph for the [arduino-mbed-bridge](https://github.com/sparkfun/arduino-mbed-bridge/network) repo we can find the `25fe1981fe509217059f093395372e1ffcbd6335` lives on the `remove-api-submodule` branch:

[image]

If we have a look at the [commits](https://github.com/sparkfun/arduino-mbed-bridge/commits/remove-api-submodule/) we can see that commit `25fe1981fe509217059f093395372e1ffcbd6335` is the second most recent.

The [commit message and code diff](https://github.com/sparkfun/arduino-mbed-bridge/commit/b239f175e5fa3c5855f685be4690d93e694abf5a) of the most recent commit [`b239f175e5fa3c5855f685be4690d93e694abf5a`](https://github.com/sparkfun/arduino-mbed-bridge/commit/b239f175e5fa3c5855f685be4690d93e694abf5a) on the branch `remove-api-submodule` would indicate the @Wenn0101 believes he inverted the logic of the interrupt functions:
[image]

paulvha, 2024-03-26:
Hi Nigel,

There has been a modification done later on :https://github.com/sparkfun/Arduino_Apollo3/issues/278.
That was done by GIGAPOD on 23 Aug 2023, BUT that was only posted as a pre-release as mentioned in the issue.

Could that be the reason?

regards,
Paul
Greetings from Amsterdam.

image.png (view on web)<https://github.com/sparkfun/Arduino_Apollo3/assets/365332/8bf65c96-2914-441c-a451-da5fe5aec77b>

image.png (view on web)<https://github.com/sparkfun/Arduino_Apollo3/assets/365332/4e31c93e-2171-42b7-aed8-1a0d61e5278b>