sparkfun/Arduino_Apollo3 #481: Setup development environment in VSCode for Ambiq microcontroller

RuXBee, 2023-04-03:
## Subject of the issue
@RicoFio I am wondering to know how you setup a development environment in order to flash and debug program on Ambiq microcontroller in VSCode.
Coudl you share with me the *.vscode* folder?
I have read your [issue #36](https://github.com/sparkfun/Arduino_Apollo3/issues/83)
I would like to develop bare-metal firmware with the Ambiq SDK (HAL libraries), but inside the VSCode.
Did you find the SVD files in order to see the peripherals register?

Thanks in advance for any idea that you bring to me  😃

justiceamoh, 2023-04-04:
Hi @RuXBee, 

Yes, I did my firmware dev in VScode for the Artemis board. Find attached a project template, including the `.vscode` folder.

The SVD file is provided in the SDK: `AmbiqSDK/pack/SVD/apollo3.svd`

[artemis_cpptemplate.zip](https://github.com/sparkfun/Arduino_Apollo3/files/11148332/artemis_cpptemplate.zip)

RuXBee, 2023-04-04:
Great, I got it.
Thanks very much!

closed 2023-04-04 by RuXBee