sparkfun/Arduino_Apollo3 #488: Servo Library Inoperable on LoRa thing plus Explorable

e-deck, 2023-10-29:
The servo library does not function on the Sparkfun Explorable board.
Servos are verified to function correctly using analogeWrite, but issued writes from the Servo library simply do nothing.
There is no output voltage on the signal pin.
Again, pin works fine. Servo works fine. I tried on multiple pins too.

paulvha, 2023-10-29:
It might be the pin-to-pad numbering.

what happens if you use myServo.attach(D8); instead of myServo.attach(8); (so an 'D' in front in front of the pin).

e-deck, 2023-10-29:
> It might be the pin-to-pad numbering.
> 
> what happens if you use myServo.attach(D8); instead of myServo.attach(8); (so an 'D' in front in front of the pin).

Yeah, this worked. Wow, I spent so long looking up and down trying to figure out what was wrong, making COMPLETELY sure it was the library and not something I screwed up. Well hopefully now anybody else who has the same problem will stumble on this and won't waste as much time as I did. Thanks for the quick feedback. You're a lifesaver. Hopefully, this gets patched soon.

paulvha, 2023-10-29:
Good to hear this works. With the early versions of the Artemis boards the PIN and PAD numbers were the same. In later boards that is not the case, but indeed the examples should be updated to reflect that.