sparkfun/Arduino_Apollo3 #497: 403 error when downloading arm-none-eabi

gauteh, 2025-02-28:
STEP 8/28: RUN  arduino-cli core --additional-urls https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/refs/heads/main/package_sparkfun_apollo3_index.json install "Sparkfun:apollo3@1.2.1"
Downloading packages...
SparkFun:arm-none-eabi-gcc@8-2018-q4-major Server responded with: 403 Forbidden
Error during install: Server responded with: 403 Forbidden
Error: building at STEP "RUN arduino-cli core --additional-urls https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/refs/heads/main/package_sparkfun_apollo3_index.json install "Sparkfun:apollo3@1.2.1"": while running runtime: exit status 1

gauteh, 2025-02-28:
❯ wget https://static.sparkfun.com/large/gcc-arm-none-eabi-8-2018-q4-major-mac.tar.bz2
--2025-02-28 11:15:41--  https://static.sparkfun.com/large/gcc-arm-none-eabi-8-2018-q4-major-mac.tar.bz2
Resolving static.sparkfun.com (static.sparkfun.com)... 52.5.46.38, 54.84.188.200, 2600:1f18:41d4:e9fb:fae1:f2e7:abf2:6e0b, ...
Connecting to static.sparkfun.com (static.sparkfun.com)|52.5.46.38|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2025-02-28 11:15:42 ERROR 403: Forbidden.

PaulZC, 2025-03-01:
Hi @gauteh @jerabaul29 ,

Apologies. We are (now) aware... This was reported on the [Forum](https://community.sparkfun.com/t/arduino-apollo3-server-problem/63643) too. The issue should be fixed. Please try again.

Thank you,
Paul

PaulZC, 2025-03-01:
Resolved by #498 . Apologies for the inconvenience! Please reopen if you see any further issues.

Thank you,
Paul

closed 2025-03-01 by PaulZC